mjuhanne/fyrtur-motor-board #1: Src\motor.c:594:uint8_t check_voltage()  never called

arturdd, 2021-01-06:
Hello
I LOVE your work on this device.   It's a piece of art :)
For me, making Fyrtur silent is the most important, missing feature of this device

As a first step, I'd love to change the motor firmware, to adjust default speed, as it's extremely noisy in the morning.
So it will be working with Zigbee module for now. ESP32+DC mod is planned for some time in future

It appears I need to modify  value of DEFAULT_TARGET_SPEED to 3 (RPM) for silent work
And also DEFAULT_MINIMUM_VOLTAGE to some real value, like 6 ? (volts)

During code review I've found that function check_voltage() exists but is never called and I am wondering if that matters in case of Battery operations.

Also could you please, share some advice's regarding IDE settings -  in terms of compiler optimizations etc ?

Thank you

mjuhanne, 2021-01-07:
Thank you for your kind words :)  

Damn, the check voltage function is one of the things I overlooked since I've been using DC mod from the beginning. I even wrote all the **minimum voltage** related functionality (e.g. writing the value to EEPROM) but forgot to test it. It will be easy to call check_voltage() from motor_process() though.

That being said, I would advise you to use the DC mod anyway, since there's no sleep functionality implemented and the STM32 and the motor board peripherals will continue to draw many milliamps continously and will empty your battery in no time (few days I think). I will have to implement that in the future. I have to confess that this Fyrtur project is the first one I'm developing with STM32 so everything is new to me as well.  A bit of googling revealed that HAL_SuspendTick() along with HAL_PWR_EnterSLEEPMode() should do the trick. Also peripherals need to be powered down with PWR_EN pin.  A wakeup interrupt has to be set to in order to continue when message is received via UART. Testing needs to be done to make sure the 1st bit is received correctly as well so the waking message won't be corrupted.

Your assumptions regarding DEFAULT_TARGET_SPEED and DEFAULT_MINIMUM_VOLTAGE are correct. I'm eager to hear how you will manage with the Zigbee module. In theory it should work since I tried to emulate the original firmware functionality as best as I could, but I never got to test it since I converted all the blinds to use ESP32/8266 already a while back.

Good luck :)

genehand, 2021-05-09:
Chiming in to agree that is an impressive and unexpected project.  I flashed it today then found this thread, so I put it back to stock for now.  The battery is much more convenient for me, the blinds are on a balcony door that would be messy to run power to. I'll watch the repo to see if someone is able to add those sleep settings in.

Side note, I have a ST-Link v2 compatible stick like [this one](https://www.amazon.com/DAOKI-ST-Link-Programming-Emulator-Downloader/dp/B01EE4WAC8/) and had to use Hot Plug mode in the ST-Link utility to detect the target.

mjuhanne, 2021-09-19:
@genehand Just a heads up. Now the recent firmware version has sleep mode support :)

arturdd, 2021-09-29:
> Side note, I have a ST-Link v2 compatible stick like [this one](https://www.amazon.com/DAOKI-ST-Link-Programming-Emulator-Downloader/dp/B01EE4WAC8/) and had to use Hot Plug mode in the ST-Link utility to detect the target.

No need to fight with the official STM32-St-Link app
These cheap  ST-link, compatible sticks works perfectly well with the Open Source ST-Link utility.

tracetechnical, 2022-02-02:
> Chiming in to agree that is an impressive and unexpected project. I flashed it today then found this thread, so I put it back to stock for now. The battery is much more convenient for me, the blinds [...]

Also, the newer STM32 Cube programmer from ST also works! I had no luck getting the ST-Link utility to find it.

dominikkarall, 2022-03-18:
I tried to enable battery status via the original zigbee module, but it always reports 69%. Can you please let me know which part of the code needs to be modified to enable correct battery status? Thx!

mjuhanne, 2022-04-10:
Minimum voltage checks are done in the newest version. Also battery status is now shown correctly.

closed 2022-04-10 by mjuhanne