mjuhanne/fyrtur-motor-board #2: wrong Temp / RH readings...

thncode, 2021-01-22:
This is one of the best smart home mods I have seen so far. I am currently using 6 fyrtur devices and modded them already by just using the motor and pcb using 3d printed size optimized mounts and your 8266 port is great! As I am using homekit I did another port of the software, but this is really great work, thank you!!
I have one small problem though, the sensor is delivering wrong values, always to high (35-36 degree celsius) but steady values. I searched for this effect with no success so far, did you experience something like this as well?

mjuhanne, 2021-01-30:
Thank you for your nice words and sorry for the late response.  It's always nice to hear if someone else has modded their Fyrturs. 

I'm intrigued about your setup. So are you using my PCB but with some other microcontroller? I haven't yet found any other (public) projects that control the Fyrtur motor. Please do enlighten us :)

Now that you mention, I did notice the higher temp sensor values. I just thought that it was because the sensor was in enclosure and the ESP would heat the air, but it seems that is not the issue. I will have to research this more.

thncode, 2021-02-06:
no, not another microcontroller but just different software as I am using apple homekit smarthome equipment. This allowed me to get rid of the bridge. I basically just used the motor (and now your modded pcb) as I don't like the ikea housing and 3d printed a minimalistic mounting...
[image]

simlu, 2022-06-07:
@thncode That custom printed minimalistic design looks awesome! Are you happy with how it turned out? Also do you happen to have the files handy for 3d printing? Would love to do something similar!