mjuhanne/fyrtur-motor-board #3: Motor stutters @ 50 % only when closing

hansgans007, 2021-01-30:
But when I open the blind it doesnt stutter 

Maybe u know why ? 

Can i do something because i dont want to buy a new one ?

mjuhanne, 2021-01-30:
Hi,
by stuttering you mean that it stops for few milliseconds and then continues? 

Does it happen with both the original motor firmware and my custom one? (and you are using the latest version?) 

What voltage are you using? Also, make sure that the power supply can supply at least 1 amp (preferably 2 amps).

hansgans007, 2021-01-30:
https://user-images.githubusercontent.com/76871070/106366440-1c5d5b80-633c-11eb-8bcf-76cd3792f3d2.MOV

Yes with both firmwares.

and only appears when closing ( @ 50 % ) 

From 0-49% it’s smooth 

50%-100% stutter 

from 100-0% it’s smooth also 

the powersupply makes 7.5V / 2A that’s enough.

hansgans007, 2021-01-30:
the 2ne blind don’t do it

mjuhanne, 2021-01-30:
Ouch, that clacking sound sounds quite nasty! Are you sure there's not some kind of mechanical obstruction there? 
Because there's no problem with your 2nd blind, and because the problem is with both custom/original motor firmware, then that proves it's not an issue with the firmware itself.

hansgans007, 2021-02-01:
When it drives up no Problem only @50% when it goes down 

u also dont have a solution ?

mjuhanne, 2022-04-10:
You could try the newer versions, but I suspect this is some kind of mechanical problem. Closing.

closed 2022-04-10 by mjuhanne