mjuhanne/fyrtur-motor-board #4: OTA update with deconz

popy2k14, 2021-02-14:
Hey @mjuhanne, thx a lot for this custom FW.

Currently running 7x fyrturs/kradjil with original firmware 2.2.009. They are working reliable but a bit noisy. Especially in the sleeping room this is annoying, and I want to give your firmware a try.

I am running the blinds without the repeaters in my zigbee network on deconz/Phoscon and updated all 7x with the OTAU plugin to 2.2.009 over the air.

In the Readme you are describing of flashing with STLink, this I want to avoid because of demounting and mounting.

Is it possible to flash/get '.zigbee' and '.ota.signed' files and flash them with deconz OTAU plugin without demounting the blind's?

Thx

popy2k14, 2021-02-14:
Ah, now I'll see.
That's two independent firmware/CPUs.
One for wireless module (zigbee) and one for the motor control, which you reverse engineered.

Is there any sort of bootloader on the STM32 motor module which we can use to OTA update?

mjuhanne, 2021-02-14:
Hello! 
OTA update of the STM32 motor module has been crossed my mind few times before but I haven't yet made any effort to develop it until now.

TLDR; OTA update IS possible, but unfortunately I don't know any way to force it from original firmware without first disassembling the motor module and flashing the custom firmware first via ST-Link.

-----

Firmware of STM32F030 can be updated via UART (in addition to the Serial-Wire Debug interface that's mentioned in Readme file).  But it seems that Ikea (or the manufacturer of the motor module, which I think has been outsourced) did not want to (or see the need to) have the possibility to update the STM32 firmware after initial firmware writing during assembly.    A testament to this is that the re-programming via UART requires controlling both RESET and BOOT0 pins and while the former is there on the SWD header, the latter pin is not even connected anywhere!

I proceeded by carefully soldering a wire to the BOOT0 pin of the STM32 itself, and I was able to update the firmware with ESP32 ([see here](https://github.com/laukik-hase/OTA_update_STM32_using_ESP32)), so now we know that OTA update IS possible.

Now, to make things easier, it should be possible to jump to the boot loader programmatically using custom UART command so BOOT0 pin (and consequently RESET pin as well) would not be needed. But if something happens with firmware upload, the module would be bricked since there's no way to force launching the boot loaded without those pins.  In theory one could make safeguards by modifying the ResetHandler so that boot loader is always launched if main program is not correctly run after certain timeout, but will get quite complicated fast.  The safest way would be to run the SWD wires outside module anyway (as it's done currently) in order to flash the firmware with ST-Link in case the module gets bricked.

I did make extensive reverse engineering of the original motor module firmware as well, and I did not find any commands that would hint of the possibility to update it via UART (such as jumping to the boot loader) so flashing the custom firmware with ST-Link is anyway required at least initially.

popy2k14, 2021-02-15:
Thanks for your analysis and findings.
Sad that IKEA (or/and the manufacturer) did'nt find it usefull to also add OTA for the motor module.

For me i must think about if i dissasamble all my blinds (currently they are in warranty).
Maybe later :-)

mjuhanne, 2021-03-12:
@popy2k14 FYI, the motor module firmware has been enhanced with possibility to do OTA update (currently via the Fyrtur ESP32 module). Unfortunately one still needs to disassemble the motor unit and flash a custom firmware first. Subsequent updates can be then done as OTA update.

closed 2021-03-12 by mjuhanne

popy2k14, 2021-03-15:
thx for the info and your efforts.
Sadly, i don't want to change the zigbee module to an esp32 because it works really good here (except the noise :-) )
Also don't want to disassemble it during the warranty.
Maybe i will open/flash the blinds at a later point int time :-)