mjuhanne/fyrtur-motor-board #5: Modify speed only

Jellyfrog, 2021-04-11:
Hi

I'm only interested in changing the motor speed (like others it seems)
And I read in another issue that this firmware will drain the battery currently.
So my question is if it's possible to modify only the motor speed of the original image with some hex editing?
(I know they are signed but maybe that only applies to OTA?)

Any advice?

Jellyfrog, 2021-05-01:
@mjuhanne any tips?

popy2k14, 2021-06-03:
+1 thats the only change i need -> chaning motor speed and maybe acceleration.

Dont have (and want) wires to the my 8x blinds, i am happy with the battery life and also zigbee function (yes there are issues sometimes, but they are rare and i can live with that but satay wireless :-) )

So for we also the only relevant thing is changing the motor speed to a lower value so i wake up from sunlight rather than whining of the blind/motor.

Was thinking about using this cutsom firmware after waranty expires but sadly read about battery drain.

@mjuhanne Any hint we can help to solve the battery drain or maybe @Jellyfrog idea to hex edit the original motor firmware?

Can you please help us for a better wakeup experience :-)

mjuhanne, 2021-09-19:
@Jellyfrog  and @popy2k14 . Sorry for the delay, I have been quite busy with other things for the past few months :)

Please see the updated firmware version. It has sleep mode programmed, so you should be able to use it also with battery installed.  There are two precompiled binaries in the bin/ folder so you should check them out (one with sleep mode on and one without).  

Also there's a new feature which I dubbed "Flexi-Speed" which allows us to change the motor speed even with original Zigbee module. Just try to rewind the curtain up three times in a row when the curtain is already at top position. This signals the Flexi-speed code that we want to change speed from one speed 'preset' to another (there are 4 different presets: 3,5,15,25 RPM).  To be honest, I haven't tried this on original Zigbee module since I've already converted all of them to my own design. Feel free to test them but unfortunately I will not be able to give any additional support for this feature. If there are any problems, we can of course try to figure them out..

arturdd, 2021-09-29:
This is Amazing !!  You are my god & savior !!

Jellyfrog, 2021-09-29:
It's indeed very cool. One day I promise I will try it. Need to buy a flasher tho :)

joBr99, 2021-10-10:
Just soldered cables to the motor board, will give it a try after the St-Link Debugger has arrived.

joBr99, 2021-10-12:
I've to admit that i've flashed your firmware without backup of the stock one, worked out fine everything is still working and setting RPM Speed works as intended. I will give the open source version a try on the next one. (and yes my StLink is also a China Clone)

I connected the Cable to the Zigbee Module for the supply of the external power (so rx/tx was also connected)

popy2k14, 2021-10-14:
@mjuhanne 
Flashed the bin/fyrtur-0.82-sleep.bin firmware on the fyrtur in my bedroom.
I have an original ST-LINK v2 (updated to v...37 with utility) and reading with newest ST-LINK Utility was no problem (powered by the battery over the zigbee board).
Sadly i could'nt flash with ST-Utility.
The error was: Error occured during program operation!

After using the recommended STM32CubeProgrammer, flashing worked like a charm.
Maybe you can add this information to the readme?

I have set the RPM to 3 (i think) because i can barley hear the blind and all is working with the original unmodified zigbee module. Hope the battery will last as long as with the original FW.

A huge thank's to eliminate the whining in the morning :-)

Keep up the good work

popy2k14, 2021-10-15:
@mjuhanne 
Modded my second fyrtur in the bedroom of my son.
Sadly with this blind the CFW doesnt work well.
The following issue's occured:

- Response to the zigbee module button presses are "crazy". Sometimes there is no reaction, sometimes the wrong one (pressing up -> blind goes down). Also setting the flexi speed and end position in nearly impossible.
- Sometimes there is a sleep of death, removing battery resolves this
- Readings of position is messed up -> blind is 10% down and read's 100%. Next button press down it goes all the way down... and so on.
- One time it passes the curtain length and rewinded it itself until i removed the battery
- The Motor is ticking "loud" in an intervall also on low speeds -> already checked and assembled it again. Maybe that's just a bad motor.

I have done the following to resolve the issue:

- Reflashed v82_sleep incl. chip erase -> same issue
- Reflashed the backup -> blind is working as expected again

Any hint's what's wrong? 
In my bedroom the CFW is running well.

thx

joBr99, 2021-10-16:
Just flashed another two of my blinds, I was able to read out and backup the original firmware, after changeing Readout Protection Level from 1 to 0 with the open-source commandline tool. 

@popy2k14 Could not see any abnormal drain of the battery.

mjuhanne, 2021-10-17:
@popy2k14 , @joBr99   Nice to hear you've finally gotten a working setup!  I may have a idea what has been causing the erratic functionality of your other blinds.  If you are running blinds with lover voltage source than nominal 7V or the blinds have higher friction than normal (as I suspect it is because you said it is causing some weird sounds), the motor might draw higher current than allowed (this is set via DEFAULT_MAX_MOTOR_CURRENT and its default setting is 1024 mA).  If this limit is reached and blinds were opened, the firmware assumes that top position is reached and sets its position as 0. If closing curtains and max current draw is reached, there will be error flag set but position is kept unchanged.  

I could see this behaviour on one of my blinds when they were running on 5V voltage source and were kept idle for a day or so (so the static friction was settled high). When attempt was to open the blinds, nothing happened but position was set to 0. Then trying to close blinds could cause them to wind over maximum length eventually.  

If this behaviour is continuing, you could try one of the following:
1) Higher voltage source
2) Configure the max motor current to 2A  (recompiling it with DEFAULT_MAX_MOTOR_CURRENT set to 2000. Manually setting the maximum current is also possible but required ESP32)
3) See if something causes additional friction and try to get rid of that

popy2k14, 2021-10-17:
@mjuhanne 
thx a lot for the info and analysis.
That makes totally sense! Since it just come up on one of the blinds.

Any idea what's the default DEFAULT_MAX_MOTOR_CURRENT setting in mA in the default firmware?

As you said you also seeing this issue on one of your blinds, maybe setting to an higher value for default would be good?

I now have two blinds with your CFW running and hoping to not get any of the issues with those.
One the one which makes the issue, i have double checked everything on the mechanical side (aluminium tabs ...) all seems good. Maybe the motor self draws more current than on the others.

Thanks

mjuhanne, 2021-10-17:
@popy2k14 The default max current is 1024 mA, but now I realized that if you want to change the default settings (but cannot use the ESP32 interface to change those on-demand) , then simply modifying the source code, recompiling and uploading the FW is not enough because the default settings are saved to the flash memory on first run and then subsequently loaded from there.  Normally FW uploading overwrites only the code section (lower flash address ection), so you need to clear the whole flash memory when uploading the modified FW.

joBr99, 2021-10-17:
Yes, I did flash my fyrtur's with the battery connected, this works just fine.

joBr99, 2021-10-17:
Just checked my two firmware dump's I did with the cli version, they are also garbage :/

[image]

mylesagray, 2021-11-05:
Has anyone tried setting a custom max curtain length in addition to the custom speed?

Like the others, Zigbee and wireless is "good enough" for me, but my main bug-bear with the blinds is that I have a 205cm drop and even if I changed the material, the blind will only go to 195cm.

popy2k14, 2021-11-05:
That's an issue i faced. I could not reset max curtain length once set with this firmware and the zig bee module.
@mjuhanne said earlier that you have to erase the the complete chip (chip erase in ST tool) and flash the firmware again.
Did'nt treid this because had other major issues with this specifix blind. I just flashed my backup and used another blind and set the sorrect length.

Maybe you can try this?

@mjuhanne 
Can you please look into the issue that long pressing booth buttons for about ~10 seconds reset the zigbee module but not the max curtain length. Here is a video: https://www.youtube.com/watch?v=1b3j39tYcgc

mylesagray, 2021-11-05:
@popy2k14 so you managed to create a drop >195cm with this firmware and the original zigbee module?

I'm waiting on my ST flash tool to arrive, so i'm just getting stuff lined up.

popy2k14, 2021-11-05:
@mylesagray 
Sorry, i think i got you wrong. No i did not. I thought your issue was you cant reset the max curtain position, but you ant an "longer" curtain. Sorry, no experience here with longer curtains.

popy2k14, 2021-11-11:
@mjuhanne
Experienced the next "issue" with one of the flashed devices (v82_sleep version + zigbee module).

This now happend two times:

- In the morning my smarthome sends and position cmd to open it ~1/3
- Today i was awake and it moved a few cm up and just stopped with an louder/hearable "click"
- Then it just stops
- To resolve this yesterday, my wife said she just pressed "up" on the ikea remote (which is also connected to my zigbee network)
- To resolve this today i also pressed "up" but nothing happend, counted to 10 and pressed "up" again -> now the blind was moving to the top, but faster as the usally set speed. I think vari speed changed the speed.

So my assumptions/quedtions are:
- not the current limit exceeded 1024mA the first time, because varispeed not changed!?
- Is it possible that with an faster "varispeed" that never happens, because the motor/curtain has some mechanical momentum?

Can you explain this behaviour? (you have mor insights and understanding on the software :-) )
I am close on reverting to the original firmware because of those issues.

Hope you can help me, it's so nice quiet with this CFW.
thx a lot

Jellyfrog, 2021-11-18:
Does someone know if it would be possible to read/flash via gdb?
Ive flashed another trådfri product using https://github.com/walmis/blackmagic-espidf, this was however a `.s37` file which seems to work a bit different.

Jellyfrog, 2021-11-23:
So to confirm, I should connect both the 3.3V and external(battery) *at the same time*? Feel a bit weird

popy2k14, 2021-11-23:
Yes, correct.

dominikkarall, 2021-12-20:
Hi,
many thanks for the great work!
I installed the motor cfw today and it works excellent. Would it be possible to add a morning mode with 1RPM or even lower as well? I was thinking about opening the curtain for 15min. Would that reduce or increase battery drain?

popy2k14, 2022-01-15:
Dont think that such low speeds are possible, but maybe @mjuhanne can answer this.

@mjuhanne had the issue again that the blind thinked that it is at top position, but it was at the bottom after an Position command from an smartphone. 

So maybe the current Limit exeeded the too low threshold of 1024 mA. 

Is it possible to get an zigbee sleep Firmware with the 2048 mA you have statet above? 

Thanks

arturdd, 2022-02-18:
> @Jellyfrog and @popy2k14 . Sorry for the delay, I have been quite busy with other things for the past few months :)
>
> Please see the updated firmware version. It has sleep mode programmed, so you [...]

Hi @mjuhanne  
Could you please, verify you last commit was complete? I've compiled your source code recently, with #define IDLE_MODE_SLEEP_DELAY  in main.c line  121. Unfortunately, as soon as device goes into sleep, (after 3 seconds) & wakes up after a button press (I am still using the original zigbee module) quite often the roller moves in opposite direction than requested. I've erased the flash, before flashing with my firmware, with no luck. Your compiled binnary file is free of this unexpected behavior. I've spent a few hours, debugging, also re-downloading the source code & compiling with no success.
Maybe there is something that needs to be reconfigured for sleep mode, in the device configuration file fyrtur-motor-board.ioc ?

mjuhanne, 2022-04-03:
> Hi @mjuhanne Could you please, verify you last commit was complete? I've compiled your source code recently, with #define IDLE_MODE_SLEEP_DELAY in main.c line 121. Unfortunately, as soon as device [...]

I wish I could test this with original Zigbee module. Could you test the new versions? Disabling the maximum current sensing (v0.83) and disabling debugging LEDs when entering/exiting sleep mode and thus preventing extra 2 seconds delay during commands (v0.84) could help possibly.

mjuhanne, 2022-04-05:
@Jellyfrog @popy2k14 @joBr99 @arturdd 

So it seems I was barking the wrong tree and the motor current isn't the culprit after all! :)  Instead it's the motor stalling because of low voltages. At 2 RPM there' a duty cycle as low as 15% which causes effective voltage of 1.15V..
With v0.85 the stalling should be less of a problem and even if it occurs, there's no position mismatch.

See below:

----

Versions 0.83 and 0.84 had the maximum motor current limit disabled because it was postulated that this 
was causing false detection of motor stalling. This however seems no to be the case, so from 0.85 the limit is re-enabled with a maximum current of 2A. 

Problems with motor stalling usually occur with lower speeds (< 5 RPM) and the main reason this is most likely because the motor not designed to be used with such low voltages associated with these speeds. For example with 2 RPM the PWM duty cycle causes the effective voltage to be 1.2V - 1.8V which makes the motor more suspectible for stalling should there be any extra friction during movement. 

Normally when motor stalls the system assumes the curtains are winded up for calibration and resets its position to zero. However if the motor stalls prematurely on lower speeds, there's a position mismatch which might later cause movement of the curtains outside the physical lower limit and eventually rewinding of the curtains in the opposite direction. This could cause annoying mess with wrinkled curtains and even damage the gluing between the curtains and the axel rod!

Few changes were made to 0.85 to make premature stalling less likely or at least the recovery from this more robust. The most important is a new feature called MINIMUM CALIBRATION CURRENT (by default this is 500 mA). In my tests I've measured currents during normal calibration (when stalling at the top) to be around 1-2A depending on the motor speed, where as the problematic stalls occur usually with less than 300 mA. 
So if the motor stalls at levels lower than the minimum calibration current, it is assumed that the curtains aren't really
at the top position and the position is not reset in this case. Even though the motor has stopped it's correct position is 
still known.

If these problematic stalls happens, you can try using a higher speeds or try to make sure there isn't any extra friction in the system (such as those protruding metal tabs in the motor housing when it was disassembled and reassembled). In any case,
it seems that 3-4 RPM is the minimum speed which the Fyrtur can sustain without stalling but your mileage may wary.

arturdd, 2022-04-07:
As v0.83 & v08.4 did not solved the motor stalling issue,  I've modded my Fyrtur & added an abec bearing to the mount on the opposite side to the motor. Bearing reduced this issue a little for low RPMs but did not resolved it. Here is a link to the bearing mount adapter:  https://www.thingiverse.com/thing:5342248

Yesterday I've flashed v0.85 (with sleep disabled) & I  ❤️  it. I can confirm that v0.85, at least with the bearing, works flawless at 3 RPM.  I've been testing it for a few hours 😃 and not a single fail. Thank you @mjuhanne  !! You are a real hero !!

popy2k14, 2022-04-07:
@mjuhanne NICE TO HAVE YOU BACK!!!!! and thanks a lot for your work. In the meantime i had one blind which had an defect motor and got replaced. As you saw from my other PR & posts. i had much trouble with v0.82. So your findings and changes seems that they solves the issues. But i dont know if i flash them because of not so much freetime.

One thing i saw during my tests with 0.82 is that the led isn't showing anything, is that normal?

@arturdd 
Are you running the latest 0.85 sleep versions with the original zigbee module?

arturdd, 2022-04-07:
@popy2k14 Yes, I've tested v0.85 with sleep - unfortunately my unit, is not 100% stable with the zigbee module. Occasionally, commands are ignored & you have to press the button again. Same with the commands sent by the  smart home over the zigbee. But unexpected stops of the motor appears to be fixed. Overall, I encourage you to flash your blinds with v0.85  - works much better. 

Some time ago, I've switched from battery to the DC, and firmware v0.85 with sleep disabled,  works great.

mjuhanne, 2022-04-07:
@arturdd @popy2k14 
Artur, nice to hear that you got it working :)

I went back to Ikea for some swedish meatballs and another Fyrtur blinds so I could test CFW better with Zigbee module. When snooping the UART traffic between my firmware and Zigbee module, I can confirm that there's some funky stuff going on when waking up from sleep mode.  The first message from module is not received correctly and also the reply is garbled. After this the communication is normal. This is something I haven't seen with the ESP32 module so I have to investigate further. 

I want to give heads up also for the upcoming version, so @popy2k14 maybe can wait a bit before re-flashing to firmware. 
In my testing version the blinds can continue movement after stalling (in v0.85 the position was not lost but movement stopped), so those problems should be over now. Also whenever stalling occurs the firmware increments the minimum speed by 0.25 RPM so user can flash for example the 2 RPM version and eventually it settles to the slowest speed possible without hiccups. This allows us to react to the variation between different units (some might have more internal friction than others) and user doesn't have to worry about adjusting the speed.

mjuhanne, 2022-04-10:
@popy2k14  @arturdd  @dominikkarall @Jellyfrog 

Ok, so I've done extensive testing with Zigbee module and I've fixed more bugs and edge cases than I could foresee..

Flexispeed, sleep mode etc should work now. Also battery curve gives identical value to the original firmware.
Note that minimum voltage checks are done now to protect the battery (so pre-built binaries don't work if you use 5V charger).  Also for compatibility reasons those little "dance steps" (movements to acknowledge reseting blinds etc) are done full-speed regardless of the selected speed setting so don't wonder about that.

Let me know if you have any issues.

closed 2022-04-10 by mjuhanne

popy2k14, 2022-06-11:
@mjuhanne will give it another try.
Whats the difference between flexispeed & 4 or 5 rpm version?
Do they all higher the rpm by 0.25 rpm when stalled?

popy2k14, 2022-06-11:
Found it in the source. Flexispeed you can change the speed as before and with the fixed versins not.
All Versions will increase by 0,25 rpm if stal is detected.

Now running 0.87 flexispeed on two blinds and will see how it goes.
First impressions are very good!