mjuhanne/fyrtur-motor-board #10: Reset max curtain length, how?

popy2k14, 2022-01-16:
Dear @mjuhanne , thx a lot for your work!!!!!!!!

I have donloaded the source and already made an PR regarding the max current.
When i got it right. the settings incl. MAX_CURTAIN_LEN_EEPROM are stored in an "virtual eeprom" which is just the flash.

So is the only way to reset the max curtain lenght by chip erase and flash the firmware again?
or, how can i send "CMD_RESET_CURTAIN_LENGTH" from the original zigbee module?

With the default firmare you have todo an factory reset of the zigbee module to also reset the max curtain length.
Please note, i wan to use it without the ESP, just with the zigbee module.

thx

riesinger, 2022-01-29:
I have the same issue. Calibration won't work without the custom ESP32 module, which I don't really want to use (I'm just in it for the slower motor speed)

popy2k14, 2022-01-29:
@riesinger 
I have successfully changed the mac curtain length & max motor current flash setting on v82_sleep.
You can find a quick guide of the procedure here: #9 (comment)

For the MAX_CURTAIN_LEN_EEPROM the identifier is 0x5555.
The default value is hex B3 24 (which is 9395). When you set this value it's like factory reseting the MAX curtain lenght.
After changing and reflashing, you can set the curtain length again with pressing down two times after setting the desired length.

good luck

riesinger, 2022-01-29:
@popy2k14 Thanks for the tip, I will try that tomorrow :) 
I also have the issue that the motor never seems to actually register as "I'm at the top", it just continues trying forever. However, I think that's off-topic here and better discussed in #5

popy2k14, 2022-01-29:
You're welcome.
I never had that issue.
In my case, the firmware registered 'I am at the top' and it fact it was'nt.

That resulted in an state where you actually could go beyond the Real curtain length, which is bad and could damage the blind.

After Research it seams that my blind (and also one of @mjuhanne s) has a higher friction which triggered the Max Motor current. This value is default @1024 mA. I have now changed that to 2048 mA and will see how it goes.

popy2k14, 2022-04-07:
@mjuhanne as stated in the motor speed ticket, i think this is related to the lost communication data after sleep. There was no chance for me to reset the max curtain length without modifing the virtual eeprom bin file.

So i think after you figuerd out what's happening with the communication after sleep and fixed it, this could be closed.
Or in other words: The user should be able to reset the max curtain len and set it again with the buttons of the zigbee module.

thx

mjuhanne, 2022-04-10:
These issues are fixed now with 0.87

closed 2022-04-10 by mjuhanne