mjuhanne/fyrtur-motor-board #11: Wakeup only with STOP command

dominikkarall, 2022-03-20:
Hi,
I compiled v82 with sleep mode activated and it works pretty well. The only issue I have is that waking up the device from sleep mode only works when a CMD_STOP is issued. All other commands were not working when the device went to sleep mode.

I checked the source code but couldn't find something which might cause that issue. Can you please give me some advice?

The precompiled version 0.82-sleep works. So I assume that some changes in the source code are not in the github repository. Thx!

mjuhanne, 2022-04-03:
Ok, you are the second one that comments that there might be discrepancies between source code and binaries, so there must be something to it then. I'm pretty sure that I compiled and uploaded the binaries with the latest source code since they are in the same commit.  Because I haven't able to test the custom firmware with Zigbee module, there could be some issues which I'm oblivious to.  

I accidentally left some code into the previous commits (v0.82 and v0.83) in which debugging LEDs blinked 3 times when entering/exiting sleep mode and this causes ~2 second delay before the motor reacts to movements. I wonder if that could be the culprit?  Please test v0.84 where those are disabled and commands should be instantaneous.

arturdd, 2022-04-07:
For my device, any button press would wake up the device, but usually, the first command was ignored. 
I did some tests & I believe that quite often, first bytes received after wake up from sleep are corrupted. 
Therefore while motor is in a sleep state, first button press, wakes up the device from the sleep, the second button press within 3s (sleep timeout) is processed normally. 
Whats odd,  your binary file v0.82 with sleep, was less prone to this issue, than my own build of v0.82 (compiled using SMTCube) If source code is complete, maybe this is related to compiler code optimization level. Perhaps Visual Studio does better code optimization ? @mjuhanne could you please, share you compiler settings?

Overall after a few hours of testing & debbuging, I've decided to move from battery operation to DC  as this was the fastest solution :) with sleep disabled its OK

mjuhanne, 2022-04-10:
@arturdd 

Sleep mode is now fixed with newest release.  

Regarding the compiler settings, I think I've shared all the relevant project files I've got. I'm not that familiar with the build process settings nor have I changed anything when using first the STM32CubeIde and then now Visual Studio Code. 

Anyway, here's the build log for fyrtur-0.87-sleep-5RPM.bin.

```
> Executing task: "make" -j16 -f STM32Make.make  <

mkdir build
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/bootloader.d" -Wa,-a,-ad,-alms=build/bootloader.lst Core/Src/bootloader.c -o build/bootloader.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/eeprom.d" -Wa,-a,-ad,-alms=build/eeprom.lst Core/Src/eeprom.c -o build/eeprom.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/main.d" -Wa,-a,-ad,-alms=build/main.lst Core/Src/main.c -o build/main.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/motor.d" -Wa,-a,-ad,-alms=build/motor.lst Core/Src/motor.c -o build/motor.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/stm32f0xx_hal_msp.d" -Wa,-a,-ad,-alms=build/stm32f0xx_hal_msp.lst Core/Src/stm32f0xx_hal_msp.c -o build/stm32f0xx_hal_msp.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/stm32f0xx_it.d" -Wa,-a,-ad,-alms=build/stm32f0xx_it.lst Core/Src/stm32f0xx_it.c -o build/stm32f0xx_it.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/syscalls.d" -Wa,-a,-ad,-alms=build/syscalls.lst Core/Src/syscalls.c -o build/syscalls.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/sysmem.d" -Wa,-a,-ad,-alms=build/sysmem.lst Core/Src/sysmem.c -o build/sysmem.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/system_stm32f0xx.d" -Wa,-a,-ad,-alms=build/system_stm32f0xx.lst Core/Src/system_stm32f0xx.c -o build/system_stm32f0xx.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/stm32f0xx_hal.d" -Wa,-a,-ad,-alms=build/stm32f0xx_hal.lst Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal.c -o build/stm32f0xx_hal.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2  -MMD -MP -MF"build/stm32f0xx_hal_adc.d" -Wa,-a,-ad,-alms=build/stm32f0xx_hal_adc.lst Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_adc.c -o build/stm32f0xx_hal_adc.o
"/opt/homebrew/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0 -mthumb   -DSTM32F030x6 -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ff [... cut ...] Map=build/fyrtur-motor-board.map,--cref -Wl,--gc-sections -o build/fyrtur-motor-board.elf
"/opt/homebrew/bin/arm-none-eabi-size" build/fyrtur-motor-board.elf
   text    data     bss     dec     hex filename
  25160    1136    2632   28928    7100 build/fyrtur-motor-board.elf
"/opt/homebrew/bin/arm-none-eabi-objcopy" -O ihex build/fyrtur-motor-board.elf build/fyrtur-motor-board.hex
"/opt/homebrew/bin/arm-none-eabi-objcopy" -O binary -S build/fyrtur-motor-board.elf build/fyrtur-motor-board.bin
```

closed 2022-04-10 by mjuhanne

simlu, 2022-05-23:
I'm having the same issue. Sometimes only the stop command works to wake up the device. Using the current code on Github and compiling myself.

I do have a permanent power source so will try to disable sleep mode next.

simlu, 2022-05-23:
I've updated `CMSIS` and `STM32F0xx_HAL_Driver` and will see if that improves anything. The problem is sporadic, so hard to test, but I will report back here

simlu, 2022-05-26:
Unfortunately this still seems to be an issue 😔

Any suggestions what to try next?