mjuhanne/fyrtur-motor-board #14: Debugging "Internal Command Error" - what to try next?

simlu, 2022-05-15:
[image]

I'm getting an "Internal command error" when trying to connect to the motor board.

[image]

The **ST LINK V2** seems to be connected properly. I have a second ST LINK V2 that I also connected and tried. Same error.

[image]

Orange and Green are connected from the top.

[image]

Red, Yellow and Black (becomes blue when connected to the ST LINK V2) from the bottom.

I've checked the wires and solder connections by measuring resistance with a digital multimeter.

[image]

I've connected the four pin main board connector and plugged in the battery. This does give voltage to the motor board as far as I can tell.

I'm a little stuck. Any recommendations what to try next? Thank you very much for your help!

arturdd, 2022-05-16:
My st-link adapter looks the same & works
..but I've never tried to flash it with the motor disconnected from the board. If all wires are connected correctly, I suggest to assembly the motor unit.  For test purpose there no need to bend the metal tabs - warning these breaks easily.

hint #2 It's possible that you roller blind firmware is in a sleep state. Try pressing up or down button, to wake it up - the motor will start spinning - press the button again to stop the motor & reconnect right away.

kenkus, 2022-05-16:
@arturdd Just curious, do you mean you could expose the control board without bending the metal tabs?

I didn't have issues with the connection but got an error when attempting to write with ST-Link utility. Fortunately everything worked with the newer STM32CubeProgrammer software.

arturdd, 2022-05-16:
> @arturdd Just curious, do you mean you could expose the control board without bending the metal tabs? 
Nope, It's not possible to access the control board without messing with the metal tabs. 

I suggested @simlu to assembly the motor (temporary, without bending the metal bars) to re-test the st-link connection

Other alternative is to test the console version:  [Open Source ST-Link utility](https://github.com/stlink-org/stlink) 
I've tested it, worked for me (stlink-1.7.0)

mjuhanne, 2022-05-16:
I remember having some problems when flashing with sleep mode on.  I think _--connect-under-reset_  option with st-link utility helped. Also, is your st-link probe genuine? I think I've read about some problems with cheap chinese copies. The genuine V2 is dirt cheap (25-30$).

simlu, 2022-05-16:
Hello all and thank you for the quick replies! Much appreciated!

Connecting the motor itself makes perfect sense. I've tried that and am now getting a little further - the light on the st link v2 flashes for a few seconds and I get a different error: "Can't reset the core"

[image]

Has anyone seen that error before?

Regarding sleep state: I made sure to press the buttons and have the motor spin before connecting. It unfortunately makes no difference.

I've tried the `STM32CubeProgrammer`: It recognized the st link v2, but I wasn't able to connect and didn't get an error message at all.

I will try the OS version next under Linux. At the very least it might give me better debug information.

I'm not sure if my st link v2 is genuine. I'm located in Canada and am having a hard time sourcing the grey original looking st link v2. Everywhere seems out of stock or doesn't ship here. But that's something to keep in mind for sure

tldr;
- What does the "Can't reset the core" error mean?
- Next things to try: OS utility under linux and genuine st link v2

I'll get back here once I have more information. In the meantime if you have any other input / suggestions I'd be very happy to hear it!

dominikkarall, 2022-05-16:
Hi,
I would recommend using the Linux version. It never worked for me with the ST-Link Utility. I assume my st-link is a Chinese clone as it was pretty cheap :)

simlu, 2022-05-17:
All right, using the Linux OS version did it!

Interestingly to note is that there is a very brief time window after connecting the battery and the st link v2 to the usb in which the device shows up as expected. Then, as time passes, something goes wrong (maybe sleep mode?). After flashing the custom firmware this no longer happens.

So for anyone having the above issue, this is what it looks like directly after plugging in

✅ 
```
$ st-info --probe
Found 1 stlink programmers
 serial: 24004500050000433132524e
openocd: "\x24\x00\x45\x00\x05\x00\x00\x43\x31\x32\x52\x4e"
  flash: 32768 (pagesize: 1024)
   sram: 4096
 chipid: 0x0444
  descr: F0 small device
```

and this is what it looks like after some time has passed

❌ 
```sh
$ st-info --probe
Found 1 stlink programmers
 serial: 24004500050000433132524e
openocd: "\x24\x00\x45\x00\x05\x00\x00\x43\x31\x32\x52\x4e"
  flash: 0 (pagesize: 0)
   sram: 0
 chipid: 0x0000
  descr: unknown device
```

and the write failure you'd get if you tried to write now

❌ 
```sh
$ st-flash write fyrtur-0.87-sleep-3RPM.bin 0x8000000
st-flash 1.6.0
2022-05-16T20:30:02 INFO common.c: Loading device parameters....
2022-05-16T20:30:02 WARN common.c: Invalid flash type, please check device declaration
  core status: unknown
Unknown memory region
```

**tl;dr;**
There is a very brief period after connecting the st link v2 and connecting the power source (aka battery) in which the st link tool works. If you miss that window you have to replug both.

----

Thanks everyone! Really, really appreciate all your input. I'm a big OS fan and love helping when I can. So it made me feel warm an fuzzy when I saw all of you jumping in here ❤️ 

I'm just getting started with the custom firmware now, so I'm sure I'll open more threads. But this thread has served it's purpose! Thanks again!

closed 2022-05-17 by simlu

Maestroo420, 2024-08-01:
> Hi,
> 
> I would recommend using the Linux version. It never worked for me with the ST-Link Utility. I assume my st-link is a Chinese clone as it was pretty cheap :)

////
Hello! Kinda late to the party but I'm having the same problems. How can I use/try the Linux version? BIG Regards. C