mjuhanne/fyrtur-motor-board #17: Home Assistant reports position inverse

simlu, 2022-05-27:
It looks like Home Assistant reports the blind position as 100 when the blind is all the way up and and 0 when the blind is all the way down.

Maybe this is related to the weird but where the bind becomes unresponsive here https://github.com/mjuhanne/fyrtur-motor-board/issues/11

Will have to do some digging as to what is happening.

simlu, 2022-05-27:
My best guess is that there is a bug somewhere in this code https://github.com/mjuhanne/fyrtur-motor-board/blob/68bd74e97953f4e66df4a02f444c83863bfed27c/Core/Src/motor.c#L411

popy2k14, 2022-06-07:
Is this with the original zigbee module or ESP?
PS.: Planning on switching to the CFW + original zigbee module again :-)

simlu, 2022-06-07:
@popy2k14 It's the original zigbee. The only thing I did is flash the custom firmware onto the motor module.

Ive now "fixed" the blinds not responding by switching off and on the power to roll them up. That means (1) if there is a malfunction it will reset the next time the blinds roll up and (2) it has the added benefit of perfectly syncing up the roll up.

But it obviously still feels hacky. Will try and take another look at the software when I find some time

popy2k14, 2022-06-07:
ok, thx for the update.

So the inverse reporting is fixed just with an power cycle?
If that's the case: That sound's not good.
Previous tests of v82 (which was the neewest one back in the day) had some issues with stall detection and false reporting. Also spinning the blind above it's limits and reversing the blind rollup.

i know @mjuhanne has fixed multiple things in this regard.

Just want to be sure that those issues are gone before i'll give it another try,

Would be nice if you could give us an overall feedback with this setup (original zigbee moduke + CFW).

thx

simlu, 2022-06-07:
@popy2k14 No, the inverse reporting is consistent. It might be by design, not sure. Or it could be related to the front rollup option not being implemented correctly.

But there is an additional problem (which is what I was talking about), where the blinds become unresponsive. The reason why I think the issues might be related is because Home Assistant sometimes thinks the blinds are up when they are down and the opposite. In which case the next command can't be executed.

That unresponsive problem was a much bigger concern because it caused the blinds not to go down/up as expected. I have unfortunately not figured out how to reproduce. It might have to do with executing the same command multiple times.

Other than that and overall, the custom firmware works fine and customizing was relatively easy after I managed to compile it.

My experience is that power cycling is really nice to have (not sure if you're going battery or not) when the blinds become unresponsive.

simlu, 2022-06-07:
I'll post a more detailed report with pictures and instructions once I have everything fully set up here. That might give you a better idea as well of what I'm doing. Might still be a full weekends out though

popy2k14, 2022-06-07:
@simlu 
I'll looked at my 8x fyrtur connected to conbee II (deconz) -> HA.
The all report the "current_position: 100" when they are alle the way up (in my case all with it's original latest ikea firmware).
When i'll remember correctly i also was curious about this the first time i switched to HA.
Comming from fhem and there it was the opposite.
So i think this is not an/your issue.

I also had missing zigbee commands with my fyrtur's.
In my case i noticed that another zigbee router device (which is always powered) was causing issues with the fyrtur's.
After placing an hue lightstripe in between the conbee II & the fyrtur (which had issues), solved it for me.
It seems the ikea devices are a little bit picky in regards of meshing.

Another approach to solve your issue is to try an OTA to the newest image "117C-1187-23088631.zibee" which is an 2.3.088.
I updated all my fyrtur's to this version and my feeling is they respond much quicker than before.

Let us know how it goes.
Thanks for the info, so i will try the precompiled 0.87 binarys soon again.

simlu, 2022-06-07:
Sweet, thanks for all the details. Interesting point that my fyrtur original zigbee board might be running outdated firmware. How would I go about updating it? Do I need the Ikea hub for that?

simlu, 2022-06-07:
I'm running home assistant on a raspberry pie and a SONOFF usb zigbee Dongle to connect the blinds

The firmware version on the one blind I checked reports as 0x22009631

popy2k14, 2022-06-07:
You're welcome.

The firmware version 0x22009631 (2.2.009) is the previous one.
Here you can find informations about firmware files: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/OTA-Image-Types---Firmware-versions

In the IKEA table under "tradfri_connected_blind" you can see listed 2.2.009 as the newest.
BUT HA deconz addon with it's OTAU plugin using the officially ikea json file "http://fw.ota.homesmart.ikea.net/feed/version_info.json" and download's the firmware automatically.
I am using deconz + conbee II to update the firmware (it takes really long -> a whole day) but in the end it works.

Here is an python script to download the files: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/554
Or the direct link to ikea servers: http://fw.ota.homesmart.ikea.net/global/GW1.0/01.17.044/bin/10037585-5.1-TRADFRI-connected-blind-2.3.088.ota.ota.signed

But be aware i dont know the sonoff usb zigbee dongle and if it has an OTA feature nor if the provided files are compatible,

popy2k14, 2022-06-07:
PS.: SYou could also try to change the zigbee channel to not inteference with common wifi channels.
Here is a google excerp of "recommended zigbee channel":

`
To select a Zigbee channel that does not overlap with WiFi, try channel 15, 20, or 25. These Zigbee channels avoid the most common WiFi channels (1, 6, 11). Also note that channel 25 in the US has lower transmission power per FCC requirements. Third-party devices can cause intermittent wireless interference.
`

popy2k14, 2022-06-11:
@simlu I have flashed 0.87 flexispeed on two fyrtur's and can confirm that there is no difference on position reporting between the original firmware and CFW. As stated above, HA is reporting the "current_position: 100" when they are all the way up.

@mjuhanne I think this issue could be closed

simlu, 2022-06-11:
@popy2k14 👍 I'm good closing this issue. I'll open a separate one for the problem where the blind stops responding

closed 2022-06-11 by simlu