mjuhanne/fyrtur-motor-board #21: Write protection on newer fyrtur motors?

sjorge, 2023-03-27:
```[hyperon :: sjorge][~/Desktop/fyrtur-fw]
[■]$ st-flash --hot-plug --connect-under-reset write fyrtur-0.87-sleep-5RPM.bin 0x8000000
st-flash 1.7.0
2023-03-27T20:41:11 WARN common.c: NRST is not connected
2023-03-27T20:41:11 INFO common.c: F0xx small: 4 KiB SRAM, 32 KiB flash in at least 1 KiB pages.
file fyrtur-0.87-sleep-5RPM.bin md5 checksum: a0adfcbfa59fb3dd91fa69ee735dd47, stlink checksum: 0x002293ec
2023-03-27T20:41:11 INFO common.c: Attempting to write 26296 (0x66b8) bytes to stm32 address: 134217728 (0x8000000)
2023-03-27T20:41:11 ERROR common.c: Flash memory is write protected
2023-03-27T20:41:11 ERROR common.c: Failed to erase_flash_page(0x8000000) == -1
stlink_fwrite_flash() == -1```

I bought this one about 2 months ago but only now had access to an stlink/v2 ... any succesfully flashed a recently purchased motor?

faern, 2023-03-29:
I came here to report the same. I bought a new curtain just a few weeks ago. Just waited for my st-link flasher to arrive. And now I get the same error when trying to flash.

sjorge, 2023-03-29:
> I came here to report the same. I bought a new curtain just a few weeks ago. Just waited for my st-link flasher to arrive. And now I get the same error when trying to flash.

That's sad to hear, that reduces the likelyhood I somehow messed up the solder job. 😞

faern, 2023-03-29:
It's possible to unlock it again. I used this using `openocd` and I have now been able to successfully flash this custom firmware. See this link: https://techoverflow.net/2022/02/12/how-to-unlock-stm32f0x-using-openocd/

```
openocd -f interface/stlink-v2.cfg -f target/stm32f0x.cfg -c "init" -c "halt" -c "stm32f1x unlock 0" -c "reset halt" -c "exit"
```
At first this did not work. But after disconnecting the programmer and power to everything (rebooting it) I was able to overwrite the flash.

sjorge, 2023-03-30:
```
[hyperon :: sjorge][~/Desktop/fyrtur]
[■]$ openocd -f interface/stlink.cfg -f target/stm32f0x.cfg -c "init" -c "halt" -c "stm32f1x unlock 0" -c "reset halt" -c "exit"

Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.229980
Info : [stm32f0x.cpu] Cortex-M0 r0p0 processor detected
Info : [stm32f0x.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for stm32f0x.cpu on 3333
Info : Listening on port 3333 for gdb connections
[stm32f0x.cpu] halted due to debug-request, current mode: Handler HardFault
xPSR: 0xc1000003 pc: 0xfffffffe msp: 0x00000010
Info : device id = 0x10006444
Info : flash size = 32 KiB
Error: stm32x device protected
stm32x failed to erase options
```

I then removed the battery and unplugged the st-link/v2, waited for a bit and plugged them back in.

```
[■]$ st-flash --hot-plug --connect-under-reset write fyrtur-0.87-sleep-5RPM.bin 0x8000000
st-flash 1.7.0
2023-03-30T15:29:42 WARN common.c: NRST is not connected
2023-03-30T15:29:42 INFO common.c: F0xx small: 4 KiB SRAM, 32 KiB flash in at least 1 KiB pages.
file fyrtur-0.87-sleep-5RPM.bin md5 checksum: a0adfcbfa59fb3dd91fa69ee735dd47, stlink checksum: 0x002293ec
2023-03-30T15:29:42 INFO common.c: Attempting to write 26296 (0x66b8) bytes to stm32 address: 134217728 (0x8000000)
2023-03-30T15:29:43 ERROR common.c: Flash memory is write protected
2023-03-30T15:29:43 ERROR common.c: Failed to erase_flash_page(0x8000000) == -1
stlink_fwrite_flash() == -1
```

faern, 2023-03-30:
I also powered the entire thing via the battery + zigbee board when I did this flashing. So that's not what's stopping you.

Ok. I ran a whole bunch of similar unlock commands when trying to figure this out. So maybe there is some trick to this and not just the single command above. You can try googling `openocd` + stm32 unlock etc and see if you find some other commands. I don't have the links available any longer. But it should be possible in some way.

sjorge, 2023-03-30:
It's very timing based I think

```
[hyperon :: sjorge][~/Desktop/fyrtur]
[■]$ openocd -f interface/stlink-v2.cfg -f target/stm32f0x.cfg -c "init" -c "halt" -c "stm32f1x unlock 0" -c "reset halt" -c "exit"

Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.235194
Info : [stm32f0x.cpu] Cortex-M0 r0p0 processor detected
Info : [stm32f0x.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for stm32f0x.cpu on 3333
Info : Listening on port 3333 for gdb connections
[stm32f0x.cpu] halted due to breakpoint, current mode: Handler HardFault
xPSR: 0xc1000003 pc: 0xfffffffe msp: 0xfffffffc
Info : device id = 0x10006444
Info : flash size = 32 KiB
stm32x unlocked.
INFO: a reset or power cycle is required for the new settings to take effect.

Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
[stm32f0x.cpu] halted due to debug-request, current mode: Handler HardFault
xPSR: 0xc1000003 pc: 0xfffffffe msp: 0xfffffffc
[hyperon :: sjorge][~/Desktop/fyrtur]
[■]$ st-flash --hot-plug --connect-under-reset write fyrtur-0.87-sleep-5RPM.bin 0x8000000
st-flash 1.7.0
2023-03-30T15:37:58 WARN common.c: NRST is not connected
2023-03-30T15:37:58 INFO common.c: F0xx small: 4 KiB SRAM, 32 KiB flash in at least 1 KiB pages.
file fyrtur-0.87-sleep-5RPM.bin md5 checksum: a0adfcbfa59fb3dd91fa69ee735dd47, stlink checksum: 0x002293ec
2023-03-30T15:37:58 INFO common.c: Attempting to write 26296 (0x66b8) bytes to stm32 address: 134217728 (0x8000000)
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08000000 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08000400 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08000800 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08000c00 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08001000 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08001400 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08001800 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08001c00 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08002000 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08002400 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08002800 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08002c00 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08003000 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08003400 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08003800 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08003c00 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08004000 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08004400 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08004800 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08004c00 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08005000 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08005400 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08005800 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08005c00 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08006000 erased
2023-03-30T15:37:58 INFO common.c: Flash page at addr: 0x08006400 erased
2023-03-30T15:37:58 INFO common.c: Finished erasing 26 pages of 1024 (0x400) bytes
2023-03-30T15:37:58 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2023-03-30T15:37:58 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-03-30T15:37:58 INFO flash_loader.c: Clear DFSR
 26/ 26 pages written
2023-03-30T15:37:59 INFO common.c: Starting verification of write complete
2023-03-30T15:37:59 INFO common.c: Flash written and verified! jolly good!
```

Will test if it actually worked later, but I had to time both commands immediately after connecting the power.
I'm guessing before the zigbee board had time to talk to the motor or something.

But this should probably be documented somewhere that you might need to unlock it.

sjorge, 2023-03-30:
I've hooked up the bother section too, then it's easy to spot if you are to slow to start a command, once it starts spinning, it's too late.

sjorge, 2023-03-30:
Some observations on the newer motors:
- reading/writing is locked
- openocd can unlock it, but timing is a bit tricky (reading always  remains locked it seems? I keep getting all null back)

Not related but changing full drop length with the zigbee module, will need a custom binary but building with vscode works pretty well on macOS