mjuhanne/fyrtur-motor-board #24: Potential new board rev

sjorge, 2023-05-14:
[image]

I got two and flashed them about 1,5 months ago, after some protection issues I did managed to flash them. I picked up one more last Monday and finally had time to take it apart and try and flash, it looks like it's a new reversion.

The little brown cap (I think) is now a different style and the main chip hooked up to the IO headers is also way smaller.

sjorge, 2023-05-14:
ST-Link nor openocd seem to pick  up anything at all :(

Markings on the main chip:
```
CKS32F
030K6U6
NM384
2013 A
(C) CKS (02)
```

sjorge, 2023-05-14:
At first glance it looks like a cheaper replacement, I double check and re-did the soldering as I was not 100% happy with my first attempt.

openocd still didn't seem to be able to unlock it, but perhaps it wasn't locked so I attempted to flash it.

```
[■]$ st-flash --hot-plug --connect-under-reset write ../fyrtur-0.88-sleep-10RPM-full.bin 0x8000000
st-flash 1.7.0
2023-05-14T13:55:57 INFO common.c: F0xx small: 4 KiB SRAM, 32 KiB flash in at least 1 KiB pages.
file ../fyrtur-0.88-sleep-10RPM-full.bin md5 checksum: f47647ffc19bbbef59464523376dd27, stlink checksum: 0x0021ef52
2023-05-14T13:55:57 INFO common.c: Attempting to write 24896 (0x6140) bytes to stm32 address: 134217728 (0x8000000)
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08000000 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08000400 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08000800 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08000c00 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08001000 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08001400 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08001800 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08001c00 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08002000 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08002400 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08002800 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08002c00 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08003000 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08003400 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08003800 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08003c00 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08004000 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08004400 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08004800 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08004c00 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08005000 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08005400 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08005800 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08005c00 erased
2023-05-14T13:55:57 INFO common.c: Flash page at addr: 0x08006000 erased
2023-05-14T13:55:57 INFO common.c: Finished erasing 25 pages of 1024 (0x400) bytes
2023-05-14T13:55:57 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2023-05-14T13:55:57 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-05-14T13:55:57 INFO flash_loader.c: Clear DFSR
2023-05-14T13:55:57 INFO flash_loader.c: Clear CFSR
2023-05-14T13:55:57 INFO flash_loader.c: Clear HFSR
2023-05-14T13:55:57 INFO common.c: Go to Thumb mode
2023-05-14T13:55:58 ERROR flash_loader.c: Flash loader run error
2023-05-14T13:55:58 WARN flash_loader.c: Loader state: R2 0x0 R15 0x0
2023-05-14T13:55:58 WARN flash_loader.c: MCU state: DHCSR 0xA05F0009 DFSR 0xA05F0009 CFSR 0xA05F0009 HFSR 0xA05F0009
2023-05-14T13:55:58 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
2023-05-14T13:55:58 INFO common.c: Go to Thumb mode
stlink_fwrite_flash() == -1
```

It manages to erase it but flashing is not successful :(

sjorge, 2023-05-14:
Switched to a raspberry pi so I can run a newer version of stlink, I was using macOS be fore:

```
sjorge@raspberrypi:~ $ st-info --probe
Found 1 stlink programmers
  version:    V2J29S7
  serial:     53FF68065056705424410481
  flash:      32768 (pagesize: 1024)
  sram:       4096
  chipid:     0x444
  dev-type:   STM32F03x
```

It now at least identifies the chip, so that's some progress.

sjorge, 2023-05-17:
It's says V1.1 on the label (which I haven't noticed before).

I went to swap this one for one in the older style box with the oldest manufacturing date I could find in the store, it is also already the new revsion :(

2204 -> CKS chip (white pastic stopper on box, newer wall brackets with gray plastic)
2134 -> CKS chip (old style box, with older wall brackets with transparent plastic)

sjorge, 2023-05-21:
Managed to find an older style one, doesn't look like the 4 number code  on the labels have any baring to whats inside, as the lucky one had 2224 on it so newer than the 2204 one 🤷

That one flashed without issue after i unlocked it with openocd like my other ones so it definitely looks like the other style ones are incompatible. Both seem to use CKS variants of the STM32 though.

pannlala, 2023-07-14:
I was also getting the new model.
Output from a st-info --probe with a clone flasher (bought at az-delivery):
Found 1 stlink programmers
  version:    V2J42S7
  serial:     XXXXXXXXXXXXXXXXXX
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0000
  descr:      unknown device

pannlala, 2023-07-21:
After further investigation I found out tha SWD was disabled in the firmware and that was why chipid returned 0x0000.
It is possible to activate BOOT0 and then use openocd to unlock and flash.

To activate BOOT0 cut the connection between the two pins marked with a red circle. Check with an ohm meter to be sure that they are connected. The lower pin is GND and the upper one is the BOOT0 pin. I used a sharp knife to cut the PCB trace.

[image]

When the trace is open solder a wire to the BOOT0 pin:
[image]

And the other end to 3.3V (I did solder it on the back side of the pcb):
[image]

After that follow the flashing guide from @sjorge with unlocking with openocd first and then write. Do not forget to backup the original firmware first :)

<b>When flashing is done connect BOOT0 pin back to GND</b>

Thank You @mjuhanne and @sjorge for all the job on this project!

sjorge, 2023-07-21:
Wow you got it working! I ended up erm playing roulette with 2 IKEA stores to find a old one :)

This extra info should probably end up in the README.md of the main repo I think. Both the openocd unlock step and how flash the newer revisions.

pannlala, 2023-07-21:
It took me one week of trial and error and investigation :)
I started with a clone stlink/v2 programmer and thought that was the issue then I ordered an official stlink/V3 and same problem. After reading about some bluepill boards with a BOOT0 jumper I gave it a try.

simlu, 2023-07-21:
This is awesome!! Thank you for your work!

pannlala, 2023-07-22:
Is there anyone here that could compare my original firmware with an older version? I am not sure if I have broken something or if it is a new revision in other matters too but the motor isn't moving as it should....

pannlala, 2023-07-22:
I flashed back the original dump and the board is not detected by the esp. Is it possible that openocd or the dump did something to the firmware?
Do I need to protect the flash again after writing?

sjorge, 2023-07-22:
Are you sure it’s back to normal booting?I guess and easy way to test with be dumped fw and ikea zigbee board attached.~ sjorge

pannlala, 2023-07-23:
I flashed the 0.87-sleep-5rpm again and this is the esp log when pressing a button:
E (23257) BLINDS_UART: 0x3ffb3f0c   00 00                                             |..|
I (23517) BLINDS_UART: EXT_STAT: status: Stopped, current 0 mA, speed 0.00 RPM, position 50.00 pwm 0 extra 0
I (23517) BLINDS: STATUS:Moving, battery 23, 7.2V, SPD 0.00 RPM, POS 50.00 (TARGET 0.00), STEPW_REVS 0.00
I (24107) BLINDS_UART: EXT_STAT: status: Stopped, current 0 mA, speed 0.00 RPM, position 50.00 pwm 0 extra 0
W (24107) BLINDS: Blinds stopped at pos 50.00
I (24117) NODE-FRAMEWORK: Publish: /home/cover/fyrtur-esp32-a5d770/status : Stopped
I (24127) NODE-FRAMEWORK: Publish: /home/cover/fyrtur-esp32-a5d770/direction : Stopped
I (24137) BLINDS: STATUS:Stopped, battery 23, 7.2V, SPD 0.00 RPM, POS 50.00 (TARGET -1.00), STEPW_REVS 0.00
E (25247) BLINDS_UART: invalid data length 2 (expected 3)!
E (25247) BLINDS_UART: 0x3ffb3f0c   00 00                                             |..|
E (27277) BLINDS_UART: invalid data length 2 (expected 3)!
E (27277) BLINDS_UART: 0x3ffb3f0c   00 fe                                             |..|
E (29307) BLINDS_UART: invalid data length 2 (expected 3)!
E (29307) BLINDS_UART: 0x3ffb3f0c   00 e0                                             |..|
E (31337) BLINDS_UART: invalid data length 2 (expected 3)!
E (31337) BLINDS_UART: 0x3ffb3f0c   00 80                                             |..|
E (33367) BLINDS_UART: invalid data length 2 (expected 3)!
E (33367) BLINDS_UART: 0x3ffb3f0c   00 00                                             |..|
E (35407) BLINDS_UART: invalid data length 2 (expected 3)!
E (35407) BLINDS_UART: 0x3ffb3f0c   00 00                                             |..|
E (37437) BLINDS_UART: invalid data length 2 (expected 3)!
E (37437) BLINDS_UART: 0x3ffb3f0c   00 80                                             |..|
E (39467) BLINDS_UART: invalid data length 2 (expected 3)!
E (39467) BLINDS_UART: 0x3ffb3f0c   00 80                                             |..|
E (41497) BLINDS_UART: invalid data length 2 (expected 3)!
E (41497) BLINDS_UART: 0x3ffb3f0c   00 00                                             |..|
I (43257) INTERFACE: BTN(1) status changed: 1

Has other changes been done to the communication?

pannlala, 2023-08-01:
Just an update...
Forget previous message. 
I can confirm that flashing the chip was successful. I might have damaged it or some other electronics.
PWR_EN seems to be high (3.3V) constantly.
If I send message to fore move down nothing happens with the level on HIGH_1/2_GATE and LOW_1/2_GATE. HALL_1/2_OUT is both always high.

My conclusion is that the CKS32F i damaged or some surrounding electronics. I have compared all connections with the schematics. I will do some further investigation and if possible replace the CKS32F...