mjuhanne/fyrtur-motor-board #25: Documentation for modifying settings in EEPROM without UART.

tompipe, 2023-06-29:
First off, let me just express my appreciation for the work you've put into this. 

Secondly, I thought the following might be a helpful addition for those like myself, who are still using the ikea zigbee module, and not the custom one. 

As I wanted to reverse the orientation of the blind, I needed to issue the CMD_TOGGLE_ORIENTATION command over UART after flashing the firmware, but it seemed like I was unable to issue UART commands via the STLink. 

However, after reviewing the code, I figured that as the device writes the settings to EEPROM, I could program the default settings directly into EEPROM using the STLink Utility, or rather it's replacement - STM32CubeProgrammer. 

I found the settings were stored towards the end of the EEPROM, and it looked like the setting was being prefixed with the corresponding virtual addresses I found within the code. 

The orientation setting in particular was at 0x08007930 (could these be different for other devices?) and I observed adjacant memory addresses holding other settings. Based on these prefixes in the code, I guess the process would work for changing other settings: 

```
MAX_CURTAIN_LEN_EEPROM = 0x5555,
FULL_CURTAIN_LEN_EEPROM = 0x6666,
MINIMUM_VOLTAGE_EEPROM = 0x7770,
DEFAULT_SPEED_EEPROM = 0x8880,
AUTO_CAL_EEPROM = 0x9999,
ORIENTATION_EEPROM = 0xAAAA,
MAX_MOTOR_CURRENT_EEPROM = 0xBBB1,
STALL_DETECTION_TIMEOUT_EEPROM = 0xCCCC,
IDLE_MODE_SLEEP_DELAY_EEPROM = 0xDDD0
```

To change the orientation, I changed the value at 0x08007930 from 0xAAAA0000 to 0xAAAA0001, and tested the orientation was successfully reversed 😃 

Unfortunately, I've since reassembled and reinstalled the blind, so I am unable to take screenshots of the process to contribute with a PR. But hopefully this is helpful enough.

fry, 2023-09-13:
Thanks for that, I was in exactly the same situation for the max length of my curtain.

For reference, as seen in https://github.com/mjuhanne/fyrtur-motor-board/blob/main/Core/Src/eeprom.c#L344, the EEPROM starts at 0x08007800 and settings in the form of 8 bytes (4 bytes the identifiers above + 4 bytes the setting value) are added to the unwritten parts of the page as they are set.

The actual setting value in the software is then the last setting value for that specific identifier. So setting a setting to a new value without erasing the entire page would involve adding the new setting (identifier + value) at the first unwritten 8 bytes at the end. In STM32CubeProgrammer those are with a purple background.

Thanks again! :)

JoeHogan, 2024-10-18:
this is great, im trying to do the same thing, but i when i use the STM32CubeProgrammer utility to search, it cant find any data matching either 0xAAAA0000 or 0xAAAA0001 - I'm using the fyrtur-0.87-sleep-4RPM.bin file. any tips?

JoeHogan, 2024-10-18:
I think i found the settings, but i have no idea how to update them... 

[image]