mjuhanne/fyrtur-motor-board #27: Guide: how to flash the newest revision

pannlala, 2023-09-17:
After struggling a lot with the newest revision and many hours of trial and error this is the solution.

I have done this with MacOS but it should be the same on Linux. Feel free to convert the guide to a Windows friendly version.l

As the chip is read/write protected and locked one cannot backup the original firmware or overwrite without erasing the chip first. That is why the original firmware is needed. For now it can be obtained here:
https://github.com/fhempy/fyrtur-motor-board

Do not proceed without the original firmware!!!

**Step 1 prepare the PCB:**
Cut the trace between pin 31 (BOOT0) and pin 32 (GND).  Check with a multimeter that pin 31 and pin 32 is not shorted.
[image]

Solder one end of a wire to pin 31 (BOOT0) and the other end to 3.3V. On the bottom side of the PCB a 3.3V test point is easy accessible.
[image]
[image]

**Step 2 unlock the the chip**
If using an unofficial programmer one does not need to connect external power to the PCB as power will be provided by the programmer. If using the official ST-LINK (I have only tested V3) external power is needed as stated in the original instructions.

Unlock the chip with openocd (credits to @sjorge):
openocd -f interface/stlink-v2.cfg -f target/stm32f0x.cfg -c "init" -c "halt" -c "stm32f1x unlock 0" -c "reset halt" -c "exit"

Disconnect power from the board for about 30 seconds and the connect the flasher again and run:

st-info --probe

If the output is similar to this the unlock succeeded: 
 
Found 1 stlink programmers
  version:    V2J29S7
  serial:     53FF68065056705424410481
  flash:      32768 (pagesize: 1024)
  sram:       4096
  chipid:     0x444
  dev-type:   STM32F03x

**Step 3 flash the firmware:**

As the chip is read/write protected one cannot backup the current firmware nor can one overwrite the chip with cusom firmware and preserve the parameters that is not provided in the custom firmware.

1. Erase the chip with the command 
**st-flash erase**
2. Flash the chip with the original firmware that was downloaded from the link provided in the beginning
 **st-flash write original-fyrtur-fw.bin 0x8000000**
3. Flash the custom firmware of your choise
**st-flash write fyrtur-0.87-sleep-5RPM.bin 0x8000000**

**Step 4 reassemble the motor board and enjoy**

Jellyfrog, 2024-02-03:
How do I know if I have the new revision?
Mine has the bridge between pin 31 and pin 32, and it says revision 1.1 on the sticker on the back of the PCB.

Not sure if due to bad soldering or new revision :)
```
% st-info --probe
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_APIV3_GET_COM_FREQ
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_ENTER
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_DRIVE_NRST
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_DRIVE_NRST
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
  version:    V3
  serial:     001B003B3133510A36303739
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0000
  descr:      unknown device
```

Jellyfrog, 2024-02-04:
Hi I will give it a shot!
Just to double check. 3v goes to T_VCC and is used as the target voltage only, and then using the battery as power source?

Jellyfrog, 2024-02-10:
Hmm Im stuck with:

```
% openocd -f interface/stlink.cfg -f target/stm32f0x.cfg -c "init" -c "halt" -c "stm32f1x unlock 0" -c "reset halt" -c "exit"
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V3J14M5B5S1 (API v3) VID:PID 0483:374F
Info : Target voltage: 3.259535
Warn : target stm32f0x.cpu examination failed
Info : starting gdb server for stm32f0x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
```

Ive soldered PIN31 to 3.3V

JoeHogan, 2024-10-22:
I'm stuck with this error after programming with the original firmware:

[image]

hugo-laestander, 2024-10-24:
Thanks for putting together your findings. Still, I seem to have some problems.
I had an identical solder between the pin and ground, but even after following your instructions and then using the openocd command, I've not been able to connect to the board.

I get this output:
```
$ openocd -f interface/stlink.cfg -f target/stm32f0x.cfg -c "init" -c "halt" -c "stm32f1x unlock 0" -c "reset halt" -c "exit"
Open On-Chip Debugger 0.12.0+dev-01759-gedf2c82cf (2024-10-08-21:08)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V3J15M7 (API v3) VID:PID 0483:3754
Info : Target voltage: 3.277198
Error: init mode failed (unable to connect to the target)
```

```
$ st-info --probe
Failed to enter SWD mode
Found 1 stlink programmers
  version:    V3J15
  serial:     002A00343433510837363934
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x000
  dev-type:   unknown
```
I've tried both clone and authentic st-link. And ensured the correct pins too many times, while using battery of course.
Is there something I've missed, or could I have got an even newer version which new different problems?

This is the markings on my board processor.
```
CKS32F
030K6T6
NSLM9
2322 A
CKS
```
https://www.st.com/en/microcontrollers-microprocessors/stm32f030k6.html
2322 A seems to be date code for year 23 and week 22.
Might still be the same microcontroller but in slightly different format

The board also got a sticker specifying:
```
TQL25-KT(13R+265)
20230920027 9/25
```